eloquentarduino/EloquentArduino #1: Help for specific Project

happytm, 2020-01-15:
I have been reading about your library and I like how simple the code looks like using your library. I am learning coding using arduino and created some code for ESP8266 but I cannot figure out how I can use this library with my code and make it eloquent. Link to my code is below:

https://github.com/happytm/BatteryNode/blob/master/ProbeSender.ino

As you can see my code is probably best ( or worst) example which is crying for use of your library.

Please help.

Thanks.

eloquentarduino, 2020-01-15:
Glad you liked my work.
Please give me some time to carefully read your code and give you feedback.

Simone

happytm, 2020-01-15:
Thank you. Please take your time.

eloquentarduino, 2020-01-15:
See https://github.com/eloquentarduino/EloquentArduino/blob/master/examples/wip.ino for a Work in Progress file. I added some comments, started refactoring. You should be able to grasp the idea.

happytm, 2020-01-16:
Wow! This already made code very compact. Now I realize I created a mess when I wrote the code.This is very good lesson for me.

Thank you for your help.

eloquentarduino, 2020-01-25:
I added more refactoring to the code. This time I cleaned the `sensorValues()` function using class polymorphism.

happytm, 2020-01-27:
Thank you for your help and I am sorry I am late responding to your new code. I was trying to understand the new code you supplied but as newbie I  failed to understand how to implement it. I never used switch case statements before so I have to learn basic switch case usage first. 

Thanks anyway.

closed 2020-11-29 by eloquentarduino