eloquentarduino/EloquentArduino #3: How to create the cvs file

pingguoshala, 2020-02-26:
Hi, I am a beginer of the Micro ML ,Your project is super cool, I like it very mauch! 
       In th pratical of " Handwritten digit classification with Arduino and MicroML", I'am trying to create csv file order :
                  # put your samples in the dataset folder
                  # one class per file
                  # one feature vector per line, in CSV format
      But   it  aways  with a problem when Train and export the SVM classifier,I hope you offer exzample
csv files for me to study. Thank!

eloquentarduino, 2020-02-26:
I attached the CSV I generated to run the example in the post.
There's also a snippet to plot those samples, to get you an idea of what
they should look like.
Each line represents a feature vector.

Cheers
Simone

pingguoshala, 2020-02-26:
Thank to your reply, I copy the feature vector line  from Arduino Serial monitor  directly to the opend txt file, and press Enter key to  the next line, do the same  Until the last feature vector  line ,then I change the .txt file to the .csv file.and put it to the dataset folder. Is that right ？
        when I with load_features() funtion ,I meet the error 
       Traceback (most recent call last):
  File "D:/1.py", line 16, in <module>
    features, classmap = load_features('data')
  File "D:/1.py", line 11, in load_features
    samples = np.loadtxt(filename)
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\site-packages\numpy\lib\npyio.py", line 1159, in loadtxt
    for x in read_data(_loadtxt_chunksize):
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\site-packages\numpy\lib\npyio.py", line 1087, in read_data
    items = [conv(val) for (conv, val) in zip(converters, vals)]
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\site-packages\numpy\lib\npyio.py", line 1087, in <listcomp>
    items = [conv(val) for (conv, val) in zip(converters, vals)]
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\site-packages\numpy\lib\npyio.py", line 794, in floatconv
    return float(x)
ValueError: could not convert string to float: '0.00,1.00,1.00,1.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.00,0.

eloquentarduino, 2020-02-26:
Another guy had the same problem. He told me he solved replacing

np.loadtxt(filename)

with

np.genfromtxt(filename, delimiter=',', skip_header=1)

Let me know if it works

pingguoshala, 2020-02-27:
Thanks ! I   solved replacing

np.loadtxt(filename)

with

samples = np.loadtxt(filename,dtype=float, delimiter=',')

It works well.
But I  replacing  np.loadtxt(filename)  with 
 
np.genfromtxt(filename, delimiter=',', skip_header=1)

An error occurred 

Traceback (most recent call last):
  File "D:\1.py", line 24, in <module>
    features, classmap = load_features('data')
  File "D:\1.py", line 21, in load_features
    samples = np.hstack((samples, labels))
  File "<__array_function__ internals>", line 6, in hstack
  File "C:\Users\Administrator\AppData\Local\Programs\Python\Python36\lib\site-packages\numpy\core\shape_base.py", line 343, in hstack
    return _nx.concatenate(arrs, 0)
  File "<__array_function__ internals>", line 6, in concatenate
ValueError: all the input arrays must have same number of dimensions, but the array at index 0 has 1 dimension(s) and the array at index 1 has 2 dimension(s)

eloquentarduino, 2020-02-27:
Good to know, I'll update my post

closed 2020-11-29 by eloquentarduino