eloquentarduino/EloquentArduino #4: Error compiling for board Arduino Uno.

manolisstam, 2020-03-21:
greetings,
first of all congratulations for your work, you are an inspiration to all students that wants to embed ML in microporocessors.
Iam dealing with a problem as far as tha wakeword project, is concerned
i follow the instructions and create the model.h as you said.
when i compile the sketch i get an error that the model h  coulnd noc be compiled for arduino uno.
i tried with all of your sample reference models you provide still getting the same msg.
the error is as follows,
keep on the excellent work and thank you in advanced for your time
manolis

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/main.cpp:43: undefined reference to `setup'

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/main.cpp:46: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Uno.

eloquentarduino, 2020-03-22:
Please refer to the [example](https://github.com/eloquentarduino/EloquentArduino/blob/master/examples/MicromlWakeWordIdentificationExample/MicromlWakeWordIdentificationExample.ino): you have to define `setup` and `loop`, which are mandatory for an Arduino sketch.

manolisstam, 2020-03-22:
> Please refer to the [example](https://github.com/eloquentarduino/EloquentArduino/blob/master/examples/MicromlWakeWordIdentificationExample/MicromlWakeWordIdentificationExample.ino): you have to define `setup` and `loop`, which are mandatory for an Arduino sketch.

thank you for your reply
i get this error from model.h not from the sketch of the project
do i have to define loop and setup function in model.h too ?
thank you

eloquentarduino, 2020-03-22:
`model.h` is a header file that defines the functions you need for the inference. It has to be included in a .ino sketch. Again, refer to the example: it `#include "model.h"`, then calls its functions from inside the `loop`. `model.h` won't work on its own.

manolisstam, 2020-03-22:
understood,
i follow excactly what you hace done, and i get this error when the compiling process jumps int model.h file
my sketch file has the setup and loop function
i copy your sketch and your reference model and i get this error
thank you

eloquentarduino, 2020-03-22:
Please paste your .ino file here

manolisstam, 2020-03-22:
#include <arduinoFFT.h>
// uncomment when doing classification
#include "model.h"
#define MIC A0
#define NUM_SAMPLES 64
#define SAMPLING_FREQUENCY 1024
#define INTERVAL 5
#define SOUND_THRESHOLD 3

unsigned int samplingPeriod;
unsigned long microSeconds;

int32_t backgroundSound;
double features[NUM_SAMPLES];
arduinoFFT fft;

void setup() {
    Serial.begin(115200);
    pinMode(MIC, INPUT);

    samplingPeriod = round(1000000*(1.0/SAMPLING_FREQUENCY));
    calibrate();
}

void loop() {
    if (!soundDetected()) {
        delay(10);
        return;
    }

    captureWord();
    printFeatures();

    // uncomment when doing classification
     Serial.print("You said ");
     Serial.println(classIdxToName(predict(features)));

    delay(1000);
}

/**
 * Get analog readings
 * @return
 */
int16_t readMic() {
    return analogRead(MIC);
    return  (analogRead(MIC) - 512) >> 2;
}

/**
 * Get "ambient" volume
 */
void calibrate() {
    for (int i = 0; i < 200; i++)
        backgroundSound += readMic();

    backgroundSound /= 200;

    Serial.print("Threshold set at ");
    Serial.println(backgroundSound);
}

bool soundDetected() {
    return abs(readMic() - backgroundSound) >= SOUND_THRESHOLD;
}

void captureWord() {
    for (uint16_t i = 0; i < NUM_SAMPLES; i++) {
        microSeconds = micros();
        features[i] = readMic();

        while(micros() < (microSeconds + samplingPeriod));
    }

    fft.Windowing(features, NUM_SAMPLES, FFT_WIN_TYP_HAMMING, FFT_FORWARD);
}

void printFeatures() {
    const uint16_t numFeatures = sizeof(features) / sizeof(double);

    for (int i = 0; i < numFeatures; i++) {
        Serial.print(features[i]);
        Serial.print(i == numFeatures - 1 ? '\n' : ',');
    }
}

eloquentarduino, 2020-03-22:
Copy-pasted to my IDE, it compiles just fine. Try goinig step by step:

1. include all the files, define all the constants and write an empty `setup` and `loop`
2. write the functions `calibrate`, `soundDetected`, ... but keep the `setup` and `loop` empty
3. fill the `setup` and `loop` one line at a time

Tell me at which point it fails

manolisstam, 2020-03-22:
ok will do that and will let you know
the header file is suppposed to be in the same folder with the project sketch right?
thank you so much for your time

eloquentarduino, 2020-03-22:
the header file is supposed to be in the same folder with the project sketch right? Yes

manolisstam, 2020-03-22:
thank you so mach it worked .... 
compiled and uploaded 
wrong answers but i supposed it is a training issu
will solve it
thank you so much, greetings from greece

closed 2020-11-29 by eloquentarduino