eloquentarduino/EloquentArduino #6: classIdxToName never returns default state in switch

manolisstam, 2020-04-10:
Hello again, and another time congrats for your awsome ideas and works.
i/ve seen and recreate the iris and word classifications, everything is ok, BUT i never get the default switch case.
For example if i type random numbers (in iris classification ) in serial monitor i always get a prediction. 
Why is that ?
Thanx  again for your time

eloquentarduino, 2020-04-11:
What you call "random numbers" are nevertheless considered as features: there is not a "default switch case".
A classifier (all of them, not only this implemented here) will always output a class if you give it an input: it can't distinguish "real" vs "random" data.

manolisstam, 2020-04-11:
thank you, once again for your answer

closed 2020-04-11 by manolisstam