eloquentarduino/EloquentArduino #22: esp_camera_init(): Camera probe failed with error 0x103

NiKola-it, 2021-11-04:
hi,
have this error and costant reboot on esp32-camera with AI thinker camera. modified example as code:
`
#define CAMERA_MODEL_AI_THINKER
`
but i get this error:

`Camera Ready! Use 'http://192.168.4.1:81' to stream
Motion detected[E][camera.c:1379] esp_camera_init(): Camera probe failed with error 0x103
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d0d39  PS      : 0x00060330  A0      : 0x800d0fe2  A1      : 0x3ffb1e90  
A2      : 0x00000103  A3      : 0x00000005  A4      : 0x00000001  A5      : 0x0000001e  
A6      : 0x01312d00  A7      : 0x00000000  A8      : 0x800d0d39  A9      : 0x3ffb1e70  
A10     : 0x00000000  A11     : 0x00000103  A12     : 0x8008be38  A13     : 0x3ffb1df0  
A14     : 0x00000000  A15     : 0x3ffb0060  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000040  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d0d39:0x3ffb1e90 0x400d0fdf:0x3ffb1f10 0x400d23d1:0x3ffb1fb0 0x4008bc26:0x3ffb1fd0

Rebooting...
`
esp32 is working well with webserver camera example of arduino ide.

thank you for any hint!

NiKola-it, 2021-11-04:
ok, founded problem: 
` if (motion.ratio() > MOTION_THRESHOLD) {
        Serial.print("Motion detected");
        // start the streaming server when motion is detected
        // shutdown after 20 seconds if no one connects
     //   camera.begin(FRAMESIZE_QVGA, PIXFORMAT_JPEG);
        delay(2000);
        Serial.print("Camera Server ready! Use 'http://");
        Serial.print(WiFi.softAPIP());
        Serial.println(":81' to stream");
        server.start();
        delay(20000);
        server.stop();
     //   camera.begin(FRAMESIZE_QVGA, PIXFORMAT_GRAYSCALE);
        delay(2000);
    }`
commenting out camera.begin make it work.

eloquentarduino, 2021-11-05:
Glad you found the solution.
Does it works like you expected now?

NiKola-it, 2021-11-05:
i have to check the server part, when i try to access it i just get the image sent to the client.