eloquentarduino/EloquentArduino #23: Dominant color of an image

claudiovolpato, 2021-11-17:
Ciao,
prima di tutto complimenti per il tuo lavoro.
Come posso ottenere il colore dominante di un'immagine ?
Grazie

Hello,
first of all congratulations on your work.
How can I get the dominant color of an image?
Thanks

eloquentarduino, 2021-11-18:
Well, defining a color is not an easy task. 
I suggest not to work with RGB, since you need 3 values to define a color.
Working with HSV, for example, lets you use the Hue component only to define the color, Saturation and Value won't have a great impact on this.

Once you convert your pixels from RGB to Hue, you apply binning: you want to cluster the 255 (or 180) possible hues into something like "red-ish", "green-ish", "blue-ish", "yellow-ish" and so on... Even a simple bit shift of 4-5 will work fine.

Now you have an array where you want to find the most frequent item: https://www.geeksforgeeks.org/frequent-element-array/

Does it sounds feasible?

claudiovolpato, 2021-11-18:
Ciao,
grazie per la risposta.
Mi sembra una buona idea, non sono esperto di gestione immagini, proverò ad implementare una classe di gestione.
L'idea all'origine è quella di catturare l'immagine di un proiettore ed impostare la luce  ambientale più opportuna.
Avrei potuto utilizzare un sensore di colore (troppo facile), volevo sperimentare la gestione delle immagini.

Hello,
Thanks for the reply.
It seems like a good idea, I am not an expert in image management, I will try to implement a management class.
The original idea is to capture the image of a projector and set the most appropriate ambient light.
I could have used a color sensor (too easy), I wanted to experiment with image management.