eloquentarduino/EloquentArduino #26: MotionDetectionRGBExample --> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Dario-Ciceri, 2022-01-09:
Hi, I got this error in MotionDetectionRGBExample sketch. I'm using CAMERA_MODEL_AI_THINKER
Begin
17:52:11.605 -> 690 pixels changed
17:52:11.605 -> Motion detected
17:52:16.601 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
17:52:16.642 -> 
17:52:16.642 -> Core  1 register dump:
17:52:16.642 -> PC      : 0x400d1df5  PS      : 0x00060f30  A0      : 0x800d1f0a  A1      : 0x3ffb27c0  
17:52:16.642 -> A2      : 0x3ffc2af8  A3      : 0x00000000  A4      : 0x3ffc27ec  A5      : 0x000002b2  
17:52:16.642 -> A6      : 0x00000001  A7      : 0x00000008  A8      : 0x800d1de2  A9      : 0x3ffb27a0  
17:52:16.642 -> A10     : 0x00000000  A11     : 0xffffffff  A12     : 0x3ffc27ec  A13     : 0x00000000  
17:52:16.642 -> A14     : 0x000000f0  A15     : 0x00000140  SAR     : 0x0000001d  EXCCAUSE: 0x0000001c  
17:52:16.683 -> EXCVADDR: 0x00000000  LBEG    : 0x400880e9  LEND    : 0x400880f9  LCOUNT  : 0xffffffff  
17:52:16.683 -> 
17:52:16.683 -> 
17:52:16.683 -> Backtrace:0x400d1df2:0x3ffb27c00x400d1f07:0x3ffb27f0 0x400d37fd:0x3ffb2820 
17:52:16.683 -> 
17:52:16.683 -> 
17:52:16.683 -> 
17:52:16.683 -> 
17:52:16.683 -> ELF file SHA256: 0000000000000000
17:52:16.683 -> 
17:52:16.683 -> Rebooting...
17:52:16.683 -> ets Jul 29 2019 12:21:46
17:52:16.683 -> 
17:52:16.683 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:52:16.683 -> configsip: 0, SPIWP:0xee
17:52:16.683 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:52:16.722 -> mode:DIO, clock div:1
17:52:16.722 -> load:0x3fff0030,len:1420
17:52:16.722 -> ho 0 tail 12 room 4
17:52:16.722 -> load:0x40078000,len:13540
17:52:16.722 -> load:0x40080400,len:3604
17:52:16.722 -> entry 0x400805f0

Dario-Ciceri, 2022-01-09:
I think I fixed it by adding esp_camera_fb_return(frame); in loop after delay(30);

eloquentarduino, 2022-01-10:
Did you find the esp_camera_fb_return(frame); fix in the comments of the post?

Dario-Ciceri, 2022-01-10:
> Did you find the esp_camera_fb_return(frame); fix in the comments of the post?

no, I just tried because the error LoadProhibited was pointing to EXCVADDR: 0x00000000 so a NULL pointer, so I tried to "release" the camera buffer and it works

eloquentarduino, 2022-01-10:
Ok, I need to update the post to include this. You're not the first to encounter this error.

closed 2022-01-10 by eloquentarduino

tennisparty, 2022-04-13:
Hello, I have just started using this code - https://github.com/eloquentarduino/EloquentArduino/blob/master/examples/CameraMotionDetection/CameraMotionCaptureExample/CameraMotionCaptureExample.ino with an aithinker board and ov7725 camera. I am getting a very similar printout from the board after uploading:
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1324
ho 0 tail 12 room 4
load:0x40078000,len:13480
ho 0 tail 12 room 4
load:0x40080400,len:3604
entry 0x400805f0
90.91 fps
Motion detected
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400d275e  PS      : 0x00060f30  A0      : 0x800d46b4  A1      : 0x3ffb2760  
A2      : 0x3ffc2d4c  A3      : 0x00000000  A4      : 0x3ffb279c  A5      : 0x3ffc273c  
A6      : 0x00000000  A7      : 0x00000020  A8      : 0x800d2759  A9      : 0x3ffb2740  
A10     : 0x00000000  A11     : 0x3ffc2438  A12     : 0x9999999a  A13     : 0x3fc99999  
A14     : 0x7ff00000  A15     : 0x00399999  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400852a5  LEND    : 0x400852ad  LCOUNT  : 0x00000027  

Backtrace:0x400d275b:0x3ffb27600x400d46b1:0x3ffb2820 

ELF file SHA256: 0000000000000000

What is the best way to resolve this?

Thanks,
Sam

eloquentarduino, 2022-04-13:
Motion detection moved to my new blog. You should really refer to that post --> https://eloquentarduino.com/projects/esp32-arduino-motion-detection

tennisparty, 2022-04-13:
Thanks @eloquentarduino that's useful to see. Do you have an idea why I might be having that issue?

tennisparty, 2022-04-13:
I can run the first example in your blog fine, it captures images. The second example which involves the saving the image to SD card is not working. I am using the following board settings in the Arduino IDE [image]
Your blog is really useful, thanks for the code :)

tennisparty, 2022-04-13:
Strangely, it just worked and now outputs the following

Camera init OK
Best quality jpeg file size in bytes: 7428. It took 125684 micros to encode

However I am not finding any images on the SD card.

eloquentarduino, 2022-04-14:
The example on the blog saves the images to SPIFFS filesystem, not an external SD card. It *should* work by replacing

```cpp
auto file = filesystem.writeBinary("/capture_best.jpg");
```

with 

```cpp
File file = SD.open("/capture_best.jpg", "wb");
```

Not tested yet.
Of course, you have to initialize the SD card as always in the `setup` method.